OkapiLib/OkapiLib #213: Finish odometry

Octogonapus, 2018-09-22:
Odometry has been sitting around in a "basically done we just need to test it for reals this time" state. This is an issue so we can track it officially.

baylessj, 2018-09-30:
What are we looking for as a test plan exactly? I'm assuming it will be comparing the robot's assumed position from the odometry versus its actual position, but with what motor/sensor setups? I currently have a test robot with 3 encoder wheels and a test robot with 2 v5 motors, will that be sufficient?

Octogonapus, 2018-09-30:
Yes and yes. I want to see how accurate it is during controlled movements. Try to move the robots the same during all the test runs.

3038922, 2018-10-30:
Previous Odometries were purely encoder-based. I used Kalman filters to fuse encoder and gyro angles much better. Wheels always skid can affect accuracy. Hopefully there's a default fusion of GYRO and ENCODER to get angle information.

PR #380 (2019-08-24): Merge odometry branch
### Description of the Change

This PR merges the odometry feature set. This is being merged in an RC-ready condition. This PR also removes the integrated PID-related function from AbstractMotor and changes the ADI classes to use PROS' C API.

Note: This PR is not labeled as `needs docs` because the documentation for v4 is being moved to Doxygen, which generates documentation from header files. There is a separate branch which improves the documentation for this feature set.

### Verification Process

I have done manual integration testing on this branch for a few months, [... cut]
closed 2019-08-24 by Octogonapus