OkapiLib/OkapiLib #215: Investigate if AMPC needs to scale by wheel radius

Octogonapus, 2018-09-22:
`AsyncMotionProfileController` might need to scale the modified velocity targets by wheel radius instead of directly converting to a percentage of maximum velocity.

Octogonapus, 2018-09-23:
It does. See here: https://github.com/M888R/V5-Testing-Proyect/blob/testing_ampc/include/manasController.hpp

HotelCalifornia, 2018-09-23:
Above link is to a private project. If it's okay with the author, maybe you can post the relevant snippet here?

Octogonapus, 2018-09-23:
```
    void executeSinglePath(const TrajectoryPair &path, std::unique_ptr<AbstractRate> rate) override {
      currentPosition = 0;
      pathLength = path.length;
      for (int i = 0; i < path.length && !isDisabled(); ++i) {
        currentPosition = i;
        double rightRPM = mpsToRpm(path.right[i].velocity);
        double leftRPM = mpsToRpm(path.left[i].velocity);

        logger->info(std::to_string(mpsToRpm(path.left[i].velocity)) + "," + std::to_string(mpsToRpm(path.right[i].velocity)));
        set_drive_velocity(leftRPM, rightRPM);

        //rate->delayUntil(1_ms);
        pros::delay(1);
      }
    }

  bool isPathComplete() {
    return (currentPosition >= pathLength);
  }
  
  void waitUntilPathComplete() {
    while (!isPathComplete()) {
      pros::delay(50);
    }
  }

  float mpsToRpm(float mps) {
    // (meters / second) * (60 sec / 1 minute) * (1 inch / 0.0254 meters) * (1 rotation / (diameter * pi)
    float rpm = (mps * 60.0) / (0.0254 * 4.182 * PI);
    return rpm;
  }
```
The part relevant to my comment is of course calling `mpsToRpm()` and setting the results of that to the motor's built in velocity PID. Manas also found a problem with `Rate::delayUntil()` and worked around it by using `pros::delay()` instead.