OkapiLib/OkapiLib #224: Add a way to get the gyro heading in +-180

Octogonapus, 2018-09-29:
A few people have expressed interest in a way to get the gyro angle from `[-1800, 1800]`. This is a reasonable request we should support.

Neo-Zhixing, 2018-10-10:
If I remembered correctly, because the VEX gyroscope uses analog IO, its value is going to have a sudden jump from 3600 to 0 when the robot makes a whole rotation. In some circumstances it would be helpful to get the value in [-∞, +∞], so that when the robot rotates 361 degrees the return value would be 3610 instead of 10, when the robot makes 2 whole rotations the return value would be 7200 instead of 0, etc.
I'm not sure if this was already implemented in PROS core or OkapiLib?

Octogonapus, 2018-10-10:
My original assumption was wrong. The gyro value is `[-3600, +3600]`.

PR #273 (2018-11-12): Add a method to remap gyro heading
### Description of the Change

This PR adds `getRemapped()` to get the gyro value with new bounds.

### Benefits

Users can get the gyro value in `+-1800` (or other bounds) easier.

### Possible Drawbacks

None.

### Verification Process

This was not verified.

### Applicable Issues

Closes #224.
closed 2018-11-12 by Octogonapus