OkapiLib/OkapiLib #254: std::string_view and std::clamp not found

lights0123, 2018-10-26:
**Describe the bug**
When compiling my prosv5 project, i receive the error:

```
Compiling src/opcontrol.cpp [ERRORS]
In file included from ./include/okapi/api/control/async/asyncPosIntegratedController.hpp:13:0,
                 from ./include/okapi/api/chassis/controller/chassisControllerIntegrated.hpp:12,
                 from ./include/okapi/api.hpp:11,
                 from ./include/main.h:45,
                 from src/opcontrol.cpp:1:
./include/okapi/api/util/logging.hpp:29:31: error: 'std::string_view' has not been declared
                          std::string_view filename,
                               ^~~~~~~~~~~
./include/okapi/api/util/logging.hpp:55:19: error: 'std::string_view' has not been declared
   void debug(std::string_view message) const noexcept;
                   ^~~~~~~~~~~
./include/okapi/api/util/logging.hpp:57:18: error: 'std::string_view' has not been declared
   void info(std::string_view message) const noexcept;
                  ^~~~~~~~~~~
./include/okapi/api/util/logging.hpp:59:18: error: 'std::string_view' has not been declared
   void warn(std::string_view message) const noexcept;
                  ^~~~~~~~~~~
./include/okapi/api/util/logging.hpp:61:19: error: 'std::string_view' has not been declared
   void error(std::string_view message) const noexcept;
                   ^~~~~~~~~~~
In file included from ./include/okapi/api/control/async/asyncWrapper.hpp:18:0,
                 from ./include/okapi/api/control/async/asyncPosPidController.hpp:12,
                 from ./include/okapi/api.hpp:24,
                 from ./include/main.h:45,
                 from src/opcontrol.cpp:1:
./include/okapi/api/util/mathUtil.hpp: In function 'constexpr double okapi::deadband(double, double, double)':
./include/okapi/api/util/mathUtil.hpp:80:10: error: 'clamp' is not a member of 'std'
   return std::clamp(value, min, max) == value ? 0 : value;
          ^~~
```

**Steps To Reproduce**
Steps to reproduce the behavior:
* Run `make` when `#include "okapi/api.hpp"` is included in `main.h`

**Expected behavior**
Successful compilation.

**Versions**
```
PROS Project for v5 at: C:\Users\myname\testy (testy)
Name      Version    Origin
--------  ---------  -------------
kernel    3.1.3      pros-mainline
okapilib  3.3.5      pros-mainline
```

**Additional Information**
This issue occurs both on Windows with the PROS 2 toolchain (`gcc` 6.3.1) and on Ubuntu 18.04 with `gcc` 6.3.1.

lights0123, 2018-10-26:
[Here's a link](https://gist.github.com/lights0123/ff4c3807e287fbcdc8a92069252c8414) to my `common.mk` and `Makefile`. It seems that C++17 is enabled, as `string_view` and `clamp` are C++17 features.

Octogonapus, 2018-10-26:
It appears you have an old version of the toolchain. Please show the output of `arm-none-eabi-gcc --version`.

Octogonapus, 2018-10-27:
That is too old, you should upgrade to the latest version. How did you install PROS? It should have come with the latest version.

lights0123, 2018-10-28:
I installed PROS from the Pip package and using the toolchain from PROS 2, like the wiki says (I wasn't aware that PROS 3 has new installers, as they aren't mentioned on the wiki). On Linux, I used the default `gcc-arm-none-eabi` Ubuntu package, which is still 6.x. I now notice that PROS 3 bundles GCC 7.2.1.

closed 2018-10-28 by lights0123