OkapiLib/OkapiLib #321: Motion Profile Memory Permission Error

Leo428, 2019-01-18:
**Describe the bug**
In the middle of the autonomous period, sometimes when I switched the competition switch to disable, the Brain's screen will display an error box saying "Memory Permission Error"  

**Steps To Reproduce**
Create a motion profile controller, generate a path, and run the path; then disable the bot in the middle of the period. 

**Expected behavior**
The screen should not display the memory permission error. 

**Versions**
kernel    3.1.5      pros-mainline
okapilib  3.3.8      pros-mainline

**Additional Information**
```c++
profileController.generatePath({Point{0_ft, 0_ft, 0_deg}, Point{3.8_ft, 0_ft, 0_deg}}, "A");
profileController.setTarget("A");
profileController.waitUntilSettled();
```

Octogonapus, 2019-01-18:
This is a known bug. Unfortunately, I can't fix it until a feature in the kernel is implemented. For now, the workaround is to allocate all objects which have an internal task on the heap, i.e. `ChassisControllerPID`, `AsyncMotionProfileController`, `AsyncLinearMotionProfileController`.

Octogonapus, 2019-01-30:
Depends on https://github.com/purduesigbots/pros/issues/86

PR #349 (2019-02-17): Use new task notify changes to handle killing internal tasks
### Description of the Change

This PR uses the new kernel's `task_notify_when_deleting()` changes to kill internal tasks when their parent task dies. This does not directly fix the segfault problem when an internal task references a parent object on another task's stack but does fix "runaway" tasks, such as a CCPID in `autonomous()` which keeps running after auton ends. The real fix for the segfaults are the builders, which force users to allocate their objects on the heap. This way, the parent objects will never be freed out from u [... cut]
commit ef3bbae9 (2019-02-17): Use new task notify changes to handle killing internal tasks (#349)

### Description of the Change

This PR uses the new kernel's `task_notify_when_deleting()` changes to kill internal tasks when their parent task dies. This does not directly fix the segfault problem when an internal task references a parent object on another task's stack but does fix "runaway" tasks, such as a CCPID in `autonomous()` which keeps running after auton ends. The real fix for the segfaults are the builders, which force users to allocate their objects on the heap. This way, the parent objects will never be freed ou [... cut]
closed 2019-02-17 by Octogonapus