OkapiLib/OkapiLib #330: Cannot create more than 3 motion profiles

wesleycha, 2019-02-06:
### Description

I was unable to create more than 3 2d motion profile paths. The code would freeze once I tried to generate the fourth path.

### Steps to Reproduce

1. Create an Asynchronous Motion Profile Controller
2. Generate more than 3 paths

**Expected behavior:** The paths are generated successfully

**Actual behavior:** The code freezes completely once I tried to generate the fourth motion profile path.

### Versions

```
Name      Version    Origin
--------  ---------  -------------
kernel    3.1.5      pros-mainline
okapilib  3.3.8      local
```

### Additional Information

Controller initialization:
```
auto chassis = okapi::ChassisControllerFactory::create( {M_DRIVE_LF, M_DRIVE_LR}, {-M_DRIVE_RF, -M_DRIVE_RR}, okapi::AbstractMotor::gearset::green, {4.15_in, 13.75_in});

auto profiler = okapi::AsyncControllerFactory::motionProfile(
  1.0, // max 1 m/s
  2.0, // max 2 m/s/s
  10.0, // max 10 m/s/s/s
  chassis
);
```

Path generation (in `initialize()`):

```
  profiler.generatePath( // FOLLOW IT BACKWARDS
    {okapi::Point{3_ft, 4_ft, 0_deg},
    okapi::Point{5_ft, 5_ft, 0_deg}},
    "A"
  ); // this works fine

  profiler.generatePath( // FOLLOW IT BACKWARDS
    {okapi::Point{4_ft, -1_ft, -90_deg},
    okapi::Point{4_ft, 1_ft, 90_deg}},
    "B"
  ); // this also works fine

  profiler.generatePath(
    {okapi::Point{4_ft, -1_ft, -90_deg},
    okapi::Point{1_ft, 3_ft, -90_deg}},
    "C"
  ); // this also works fine

  profiler.generatePath( // FOLLOW IT BACKWARDS
    {okapi::Point{-3_ft, 4_ft, 0_deg},
    okapi::Point{-5_ft, 5_ft, 0_deg}},
    "D"
  ); // code freezes when it reaches here
```

I have tried several different paths and the issue persists regardless of the points in the path.

wesleycha, 2019-02-06:
After doing more testing, it seems that it may be an issue of running out of memory for the motion profiles. In the meantime, is there a way that I could use the same motion profile but inverted (e.g., instead of turning clockwise, turning counterclockwise?) I saw that this was an added feature in OkapiLib 4.0.0 but it doesn't seem to be available in 3.3.9.

Octogonapus, 2019-02-06:
The profiles are heavy on memory usage and users frequently run into problems when they have long paths, so I'm not surprised. Yep, looks like that feature is on v4.0.0.

Octogonapus, 2019-02-06:
I had a chance to run this myself. When you were running it, did you have the pros terminal open? My code caught the problem:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  AsyncMotionProfileController: Could not allocate left and/or right trajectories. The path is probably impossible.
```

closed 2019-02-06 by Octogonapus

wesleycha, 2019-02-07:
I didn't have the terminal open as I didn't know about the debug info available over the serial port. That lines up with the issues I was experiencing. Thanks for your help, I'll try to work out how to make my code use less memory.