OkapiLib/OkapiLib #360: AMPC's freePath function can segfault if path to free is running

baylessj, 2019-04-16:
**Describe the bug**
Calling `AsyncMotionProfileController::freePath()` to free the currently running path can cause a segmentation fault.

**Steps To Reproduce**
Set a target to the AMPC and then free the target before the path has completed.

**Expected behavior**
The controller should be disabled prior to the path's removal ideally, or the freePath function could refuse to remove the path and warn the user about the conflict.

**Versions**
PROS: 3.1.6
OkapiLib: 3.3.8

**Additional Information**
The reason I discovered this issue was because I subclassed AMPC and added a custom waitUntilSettled() implementation that allowed for exiting the movement prior to the end of the path, in case a line was detected or the vision sensor returned something useful or whatnot. In order to save space for paths I free the path soon after the path completes, and in the case of an early exit I would get a segfault when freeing the path.

Octogonapus, 2019-04-16:
Surprise! Who could have seen this coming /s
Fine. I guess I will count this as a bug. We can add an atomic flag for when a path is running and return from `freePath()` early (and log a warning) if the path is running.

nickschatz, 2019-04-29:
Instead of failing to remove the path, the more expected operation may be to stop the path running, then remove it.

Octogonapus, 2019-04-29:
I think that if a path is running and you try to remove it, that is your mistake. Maybe we need a `stop()` method you should call first, before removing a path?

nickschatz, 2019-04-30:
It looks like reset() fills that niche.

PR #366 (2019-06-29): Prevent segfault when running path
### Description of the Change
Fixes an error where a path being removed while being ran causes a segfault by preventing the path from being removed when it's running

### Motivation
Segfaults are bad

### Possible Drawbacks
This changes the behavior of removePath in a non-obvious way.

### Verification Process

Additional unit tests

### Applicable Issues

Fixes #360
closed 2019-06-29 by Octogonapus