OkapiLib/OkapiLib #364: Add derived telemetry methods to Motor/MotorGroup

nickschatz, 2019-04-29:
**Is your feature request related to a problem? Please describe.**
Determining whether a motor/group has reached its target currently requires either a wrapper class or external methods. 

**Describe the solution you'd like**
Add methods to AbstractMotor:
- double getPositionError() - difference between target position and actual position, in encoder units
- double getVelocityError() - difference between target velocity and actual velocity, in encoder units
- (possibly?) waitUntilSettled that mirrors AsyncController, which blocks until the profiled movement is finished.

**Describe alternatives you've considered**
Write a wrapper class for Motor and MotorGroup. Kind of a pain. This is really something that should be part of the API
Create the get\*Error functions for each system that requires them. This is also a pain. DRY.

**Versions**
okapilib v4.0.0
kernel  3.1.7-commit.15.40228e3

Octogonapus, 2019-04-29:
I really want to tell you that this isn't the responsibility of `AbstractMotor`, but given that it has other methods along this vein, I guess it is.

Octogonapus, 2019-04-29:
Your third point, `waitUntilSettled`, would require a `SettledUtil`, right? Unless there is a way to check if a profiled movement is done that I don't know about. If it does require a `SettledUtil`, then I don't want that on `AbstractMotor`.

nickschatz, 2019-04-29:
It would, that's why I left it as a maybe. Why don't you want a `SettledUtil` on `AbstractMotor`?

Octogonapus, 2019-04-29:
I think that would cause `AbstractMotor` to have too many responsibilities. It already has too many as-is, but that's a fault of the hardware it models.

PR #365 (2019-04-30): Add derived telemetry functions getPositionError and getVelocityError
### Description of the Change

This adds utility methods getPositionError and getVelocityError to AbstractMotor.

### Motivation

These functions make it easier to determine the state of a motion.

### Possible Drawbacks

None.

### Verification Process

Code compiles- these are simple methods and should not require additional testing.

### Applicable Issues

Closes #364 

Docs PR: https://github.com/purduesigbots/pros-docs/pull/127
closed 2019-04-30 by Octogonapus