OkapiLib/OkapiLib #421: Chassis Controller Builder tutorial uses non-existing methods

nathan-moore, 2020-01-14:
**Describe the bug**
In the "gearset and dimensions" part of the [tutorial for ChassisControllerBuilder](https://okapilib.github.io/OkapiLib/md_docs_tutorials_walkthrough_chassisControllerBuilder.html), it refers to a withDimensions function taking three arguments. Said method [does not exist.](https://okapilib.github.io/OkapiLib/classokapi_1_1ChassisControllerBuilder.html)

**Versions**
Checked on master that said overload doesn't exist.

Octogonapus, 2020-01-14:
@nathan-moore Can you please paste the code snippet here that is giving you trouble?

Octogonapus, 2020-01-25:
@nathan-moore Any updates on this?

nathan-moore, 2020-01-25:
Just double checked, and it looks like we were misreading it.

closed 2020-01-25 by nathan-moore