OkapiLib/OkapiLib #431: Chassis Skidding out with mechanum drivetrain

lerker100, 2020-01-28:
**Describe the bug**
When using chassis->moveDistance or chassis->turnAngle with mechanum wheels my robot skids out when close to destonation.

**Steps To Reproduce**

- 4 200RPM motor mechanum drivetrain
- 2 external encoders on the back axels
- Okapi V4.0.2 Pros V3.2.1

**Expected behavior**
To slow down properly without skidding out.

**Additional Information**
I notist that one of the wheels is running way two long, that locks up the wrest of the mechanum wheels.

Octogonapus, 2020-01-28:
You didn't mention if you are using the integrated control or not. The integrated control can sometimes run motors too long, past their position targets.

Regardless, this sounds like a tuning issue, not an issue with OkapiLib itself.

closed 2020-01-28 by Octogonapus

lerker100, 2020-01-28:
this is what my chassis constructor looks like 
```
std::shared_ptr<ChassisController> chassis = ChassisControllerBuilder()
  .withMotors(frontLeftMotor, frontRightMotor, backRightMotor, backLeftMotor)
  .withSensors(leftEncoder, rightEncoder)
  .withDimensions(AbstractMotor::gearset::green, {{4_in, 10_in}, imev5GreenTPR})
  .withMaxVelocity(100)
  .withLogger(
      std::make_shared<Logger>(
          TimeUtilFactory::createDefault().getTimer(), // It needs a Timer
          "/usd/Chassis_Diagnostics", // Output to the SDCard
          Logger::LogLevel::debug // Most verbose log level
      )
  )
  .build();
```

Octogonapus, 2020-01-28:
You are using the integrated control in the motors then. If you want to tune the response, then you will need to pass gains to the builder to use OkapiLib's PID control instead.

lerker100, 2020-01-28:
Ok, it's just one motor though. And I used the same code for Omni wheels and I never had a problem?

Octogonapus, 2020-01-28:
If just changing your wheels caused this, then it's definitely not OkapiLib's problem.

lerker100, 2020-01-28:
I'll try using custom gains. Does okapilib individually change the speeds on motors by more than 5% while driving?

Octogonapus, 2020-01-28:
OkapiLib never adjusts individual motors on the chassis, it always adjusts them in pairs.