OkapiLib/OkapiLib #435: Linking fails with only Pathfinder library

superrm11, 2020-02-14:
**Describe the bug**
Compiler can't seem to find Pathfinder library, but can find adiGyro just fine
`
/opt/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: bin/autonomous.cpp.o: in function pathfinder_test():
/home/ryan/workspaces/git/2019-2020-BIG/src/autonomous.cpp:90: undefined reference to d2r(double)
`

Same error message for `pathfinder_prepare`, `pathfinder_generate`, `pathfinder_modify_tank`, `pathfinder_follow_encoder`, and `pf_fit_hermite_cubic`

**Steps To Reproduce**
Call any one of the functions above

**Versions**
Pros 3.2.1 and Okapilib 4.0.3

**Additional Information**
Checked inside okapilib.a and the object files do exist for pathfinder, gcc is just unable to see them?

Octogonapus, 2020-02-14:
Something is wrong with your end of things. I'm not sure what, though. Maybe try applying this change locally? https://github.com/purduesigbots/pros/pull/207

Either way, I'm confident that this is not an OkapiLib problem.

superrm11, 2020-02-17:
I found the problem. The pathfinder libraries are not marked with extern C. Doing so fixes the linking.

Octogonapus, 2020-02-17:
They were trying to use Pathfinder's API directly; might as well reopen this.

closed 2020-02-17 by Octogonapus