OkapiLib/OkapiLib #439: Minimize turns to a maximum of 180 degrees

dcieslak19973, 2020-03-21:
**Describe the bug**
Per: https://www.vexforum.com/t/odometry-not-turning-the-correct-way/78109 it is possible for a turn to go 270 degrees clockwise when a 90 degree counter-clockwise turn would accomplish the same thing.

**Steps To Reproduce**
RPM reproduced the behavior in the thread:

What it is doing is calculating atan of (about) -.4/-.3 which is -127 deg and then subtracting the current heading of 177 (ideally it would have been 180) from it (-127 -177) or about -304. Of course odom is using the end position from the encoders so the 0-.4 and .7-1 are the ideal number which is why the log shows -307. I guess they could optimize it by looking to see if if adding 360 or subtracting 360 would result in an smaller magnitude (or any other method). 

**Expected behavior**
Per above, "they could optimize it by looking to see if if adding 360 or subtracting 360 would result in an smaller magnitude (or any other method). "

**Versions**
Name      Version    Origin
--------  ---------  -------------
kernel    3.2.1      pros-mainline
okapilib  4.0.3      pros-mainline

**Additional Information**
Though it might make more sense to implement this at the ChassisController level

dcieslak19973, 2020-03-28:
I wasn't able to make a branch or PR in this repo, so I created a fork and created a PR there.

Octogonapus, 2020-03-29:
> I wasn't able to make a branch or PR in this repo

Correct, you don't have permissions to do that.

> I created a fork and created a PR there

You have made a PR from your branch `feature/issue/DJC/#439_minimize_turns` to _your_ `master`. You instead need to target your PR at _my_ `develop`. Like this:
[image]

PR #441 (2020-04-01): Minimize turns
### Description of the Change

Changed the odometry chassis controller to minimize turns to 180 degrees (either clockwise or counterclockwise)
### Motivation

While it was suggested to do this in the chassisController, the unit tests there assumed full control over the chassis, so it made more sense to do this in the odometry chassis controller.

### Possible Drawbacks

Favors counterclockwise turns of 180 degrees.

### Verification Process

New unit tests were created to validate the functionality

### Applicable Issues

Closes #439
closed 2020-04-01 by Octogonapus