OkapiLib/OkapiLib #444: Add wrapper class for V5 IMU device similar to the adiGyro class

dcieslak19973, 2020-04-03:
**Is your feature request related to a problem? Please describe.**
Would like to be able to use the new V5 inertia measuring unit in much the same way that the old ADI Gyro was

**Describe the solution you'd like**
Similar class to the adiGyro. Perhaps with a flag that would allow one to change the X,Y,Z (yaw/pitch/roll) value returned.

**Describe alternatives you've considered**
Using the raw PROS IMU class

**Versions**
kernel 3.2.1 pros-mainline
okapilib 4.0.3 pros-mainline

**Additional context**
N/A

Octogonapus, 2020-04-04:
I am fine with this addition, but I think there should be some discussion first. What is the intent of this class (is it just to wrap the PROS API or will it do something extra)? What primary methods will the class have?

dcieslak19973, 2020-04-04:
Main thought is to allow programmers to stay within the Okapi namespace without having to dive into the pros namespace. Eventually this may allow for the Odometry to fuse the encoder readings with the IMU.

Strawman for the public methods and class name (open to feedback):

```
class V5InertialMeasuringUnit : public ContinuousRotarySensor {
  V5InertialMeasuringUnit(std::uint8_t iport, int axisType);
  virtual ~V5InertialMeasuringUnit();
  double get() const override;
  double getRemapped(double iupperBound = 180, double ilowerBound = -180);
  std::int32_t reset() override;
  double controllerGet() override;
}
```
Generally, there would just be one instance for the heading (yaw), but users may instantiate separate instances for pitch and roll. It would be sub-optimal in these cases for the user to reset pitch and roll instances in addition to the yaw, as they'd be resetting the same physical sensor 2 times more than necessary. Open to suggestions on avoidance of this beyond documentation.

Octogonapus, 2020-04-05:
> `V5InertialMeasuringUnit(std::uint8_t iport, int axisType);`

I would rather have an enum for the `axisType` instead of an `int`.

> as they'd be resetting the same physical sensor 2 times

Instead of resetting the sensor, just write to an internal offset. We will need some other method to actually reset the sensor; I suggest calling this `calibrate` instead of `reset`.

dcieslak19973, 2020-09-28:
Already done

closed 2020-09-28 by dcieslak19973