OkapiLib/OkapiLib #466: Allow users a richer experience when subclassing

dcieslak19973, 2021-02-23:
**Is your feature request related to a problem? Please describe.**

One goal of Okapi could be to allow teams to layer more complicated behaviors on top of it.  However, given that many class members and functions are `private` this can get in the way.

For example, a team may want to use Distance Sensors to build their own Odometry, rather than a tracking wheel system.  Issue #449 seeks to use the new IMU for similar purposes.

**Describe the solution you'd like**
Classes that generally favor `protected` members over `private`

**Describe alternatives you've considered**

N/A

**Versions**

4.1.0

**Additional context**

N/A

Octogonapus, 2021-02-23:
I have tried to make members `private` and `protected` where appropriate. This was a long time ago, though, so I don't remember where the problems might be. If you have concrete examples, we could discuss them.

dcieslak19973, 2021-02-24:
Sure; lets walk thru an example similar to #449 where a user wants to implement her own Odometry implementation.

After writing their implementation, a user may want `ChassisControllerBuilder` to be able to `buildOdometry` to return a chassis with their Odometry class

Octogonapus, 2021-02-24:
`ChassisControllerBuilder` wasn't intended to work that way. If you want to change that behavior, you'll probably need to do a significant amount of internal rewriting. You are welcome to try, but I would only accept it under the condition that it is non-breaking.

dcieslak19973, 2021-04-10:
Yeah, probably not worth it

closed 2021-04-10 by dcieslak19973