OkapiLib/OkapiLib #467: Allow users to set units for OdomState string representation

shbgrwl, 2021-02-26:
**Is your feature request related to a problem? Please describe.**
When logging odometry states, we can use the OdomState.str() function to do so easily. However, the function can only log the value in meters, even though it's usually easier to think in terms of feet. Although it's easy to do a rough conversion in your head, it's annoying to determine exact values.

**Describe the solution you'd like**
Parameters in the OdomState.str() function to pass in the unit (e.g. feet, inches, etc). The new function would look like this:
```cpp
std::string OdomState::str(QLength idistanceUnit=meter, QAngle iangleUnit=degree) const {
  std::ostringstream os;
  os << "OdomState(x=" << std::to_string(x.convert(idistanceUnit))
     << ", y=" << std::to_string(y.convert(idistanceUnit))
     << ", theta=" << std::to_string(theta.convert(iangleUnit)) << ")";
  return os.str();
}
```

The problem with this solution is that the unit is no longer listed in the string representation. I'm not sure if there's an easy way to convert a unit to a string representation already.

**Describe alternatives you've considered**
Users can create the string themselves if they'd like to use different units.

**Versions**
PROS: 3.3.1
Okapi: 4.2.0

Octogonapus, 2022-05-22:
@karmanyaahm WRT your note in #478:
> Another cool thing I do in my code, is printing odometry values (OdomState::str) as _tile instead of meter. 6 tiles is more intuitive than seeing 3.6576 m (the field dimensions). However, I'm unsure of how to implement that without any regressions, if you guys think it's a good idea, I can work on it with some advice.

take a look at this issue

karmanyaahm, 2022-05-23:
I can do this later this week

karmanyaahm, 2022-05-26:
So looking at this, the simplest solution here would be something like this

```cpp
std::string OdomState::str(QLength idistanceUnit, std::string distUnitName, QAngle iangleUnit, std::string angleUnitName) const {
  std::ostringstream os;
  os << "OdomState(x=" << std::to_string(x.convert(idistanceUnit)) << distUnitName
     << ", y=" << std::to_string(y.convert(idistanceUnit)) << distUnitName
     << ", theta=" << std::to_string(theta.convert(iangleUnit)) << angleUnitName << ")";
  return os.str();
}
```

And then there could be a function wrapping that.
```cpp
std::string OdomState::str(QLength idistanceUnit=meter, QAngle iangleUnit=degree) const {
    return str(idistanceUnit, QLength::getSuffixForUnit(idistanceUnit), iangleUnit, QAngle::getSuffixForUnit(iangleUnit));
}
```
Then, `getSuffixForUnit` can use `switch case` or `if else` to determine the right unit and string.
@Octogonapus what do you think about this architecture? If this works, do I need to define `getSuffixForUnit` for every type of unit or is just angle and length fine for now?

Octogonapus, 2022-05-26:
You won't be able to implement `getSuffixForUnit` in a way that has it return different values for inches, meters, etc. because that is lost in the unit storage. Everything is stored in SI units. Instead, we can let users can supply their own context-aware unit names using that longer form function.

I suggest you implement `getSuffixForUnit` so that it returns a value based on the type of the unit. E.g. `getSuffixForUnit(x)` returns `m` for `x::QLength` or `s` for `x::QTime`. Put this function inside a new file under `units/`.

The rest of your plan sounds fine.

karmanyaahm, 2022-05-26:
> You won't be able to implement `getSuffixForUnit` in a way that has it return different values for inches, meters, etc. because that is lost in the unit storage

In this case, the argument to the string function (not the value of OdomState) will be `meter` or `inch` or `tile`; it should not be any random number like 3.14_m or 6.9_ft.

So, it should be possible to use that value like the following:
```
if (input - meter < 0.0001) return "_m";
else if (input - foot < 0.0001) return "_ft";
// so on and so forth
```

Octogonapus, 2022-05-26:
Ok, give it a shot.

karmanyaahm, 2022-06-09:
Btw I have something mostly working, just give me a couple more days to refine it and PR https://github.com/karmanyaahm/OkapiLib/commit/6530761c9474f4dd183a6f6ac6a4489c7853eacd

PR #487 (2022-06-26): Allow varying units in printing odomState
Test this with:
```cpp
void opcontrol()
{
        using namespace okapi::literals;
        auto o = okapi::OdomState{x : 4.20_tile, y : 31.4_in, theta : 69_deg};

        try
        {
                printf("%s\n", o.str().c_str()); //normal, compatible with existing behavior
                printf("%s\n", o.str(1_ft, 1_deg).c_str()); // basic unit specification
                printf("%s\n", o.str(okapi::meter, okapi::radian).c_str()); // another basic unit specification
                printf("%s\n", o.str(6_tile / 100, "%", 360_deg / 100, "%").c_st [... cut]
closed 2022-06-26 by Octogonapus