OkapiLib/OkapiLib #477: Passing PID gains to an OdomChassisController affects position measurement

SomewhereOutInSpace, 2022-04-06:
**Describe the bug**
When creating an `OdomChassisController` via a `ChassisControllerBuilder,` passing in PID values via `.withGains(...)` causes significant error in position measurement, on the order of several inches. Removing the `.withGains(...)` call rectifies this error, causing the controller to report position values accurate to within <1 inch.

To provide a point of comparison, I identified this issue while empirically tuning my odometry scale measurements with regards to straight-line position tracking. My testing methodology boiled down to pushing the robot 2 feet forwards using the field tiles as a guide, then checking the X position output by the controller on the brain screen. 

Before I removed the PID gains, the robot was consistently measuring a displacement of ~17 inches, about 7 inches short of the actual measurement. Changes to track width and middle encoder distance did not have any effect; artificially inflating the tracking wheel size seemed to work, but that's hacky and probably liable to cause problems later. 

Once I removed the PID gains on a whim, the robot immediately began reporting ~24 inches of displacement. Crtl-Z'ing the PID back in caused the issue to reappear.

**Steps To Reproduce**

1. Create an `OdomChassisController` using a `ChassisControllerBuilder` with a filled-out call to `.withGains(...)`.
2. Move the robot a known distance directly forwards and compare the real displacement to the X position value output by the `OdomChassisController` instance. The controller measurement will be incorrect by a large margin.
3. Repeat the above, but without the call to `.withGains(...)`. The real displacement and controller measurement should match to within a reasonable tolerance (<1 inch.)

I'll paste my exact code in at the bottom. I _did_ try looking through the source to see how/where the gains were influencing the position calculation, but didn't turn anything up. Unfortunately I didn't have time to see if different gain values (e.g. all zeroes) influenced the level of error.

**Expected behavior**
Passing PID gains into a `ChassisControllerBuilder` when creating an `OdomChassisController` should not influence position measurement. (As far as I am aware, OkapiLib's odom motion algorithms are just standard PID loops (using the gains passed into the controller) initialized with the current state (x, y and theta) of the odometry.)

**Versions**
```
PROS Project for v5 at: /home/eagle/Documents/21417 (VEX-21417B)
Name      Version    Origin
--------  ---------  -------------
kernel    3.5.4      pros-mainline
okapilib  4.2.0      pros-mainline
```

**Additional Information**

The exact `ChassisControllerBuilder` code that caused the issue on my end:

```cpp
auto drive_train = okapi::ChassisControllerBuilder()
  .withMotors({19, 6, 9}, {12, 15, 16})
  .withGains
  (
      {0.001, 0, 0.0001}, //Distance gains
      {0.0032, 0.0003, 0.0001}, //Turn gains
      {0.7, 0.001, 0.0001}  //Angle gains
  )
  .withSensors
  (
      ADIEncoder{'A', 'B', true}, //Left encoder (inverted)
      ADIEncoder{'C', 'D'},  //Right encoder
      ADIEncoder{'E', 'F'}  //Middle encoder
  )
  .withDimensions({okapi::AbstractMotor::gearset::green, GEAR_RATIO}, {{4.125_in, 11.5_in}, okapi::imev5GreenTPR * GEAR_RATIO})
  //Specify odometry dimensions and encoder type
  .withOdometry({{2.875_in, 4.715_in, 3.5_in, 2.875_in}, quadEncoderTPR})
  .buildOdometry(); //Build an odometry-enabled chassis
```

Octogonapus, 2022-04-09:
My memory is pretty fuzzy here but from the docs it looks like you shouldn't specify both `withDimensions` and `withOdometry` when using `ChassisControllerPID` with odometry. Have you tried adjusting your `withDimensions` and then using an empty `withOdometry()`? More details here https://okapilib.github.io/OkapiLib/md_docs_tutorials_walkthrough_odometry.html

Octogonapus, 2022-05-23:
resolved offline

closed 2022-05-23 by Octogonapus