OkapiLib/OkapiLib #489: pros::controller::get_digital_new_press not implemented in okapi::controller

bnhrkn, 2022-08-04:
**Is your feature request related to a problem? Please describe.**
A method like `get_digital_new_press` isn't implemented in `okapi::Controller` which makes detecting single presses more difficult.

**Describe the solution you'd like**
Implement an equivalent method to `pros::Controller::get_digital_new_press` in `okapi::Controller`.
It should be easy to do something like:
```c++
okapi::Controller controller;
while(true) {
  if(controller.getDigitalNewPress(okapi::ControllerDigital::Y)) {
    std::cout << "Pressed\n"; // Should only print once per button press
  }
}
```

**Describe alternatives you've considered**

- Construct both okapi::Controller and pros::Controller and use either when needed
- Make a class derived from okapi::Controller which implements the missing method

**Versions**
- kernel 3.6.2
- okapilib 8.4.0

Octogonapus, 2022-08-04:
There is `changedToPressed` which implements that (though yes it's not on `Controller`). https://okapilib.github.io/OkapiLib/classokapi_1_1AbstractButton.html#a762fbcfb5065497d743a6a43fd2032f3

However, creating a button should be easy and you don't even need to assign it to a variable (I'm assuming that's why it is so cumbersome?). Use this https://okapilib.github.io/OkapiLib/classokapi_1_1Controller.html#a88c9b0063ad8455840f9ac8c24610ff0

You should be able to then do
```c++
Controller controller;
while (true) {
  if (controller[ControllerDigital::Y].changedToPressed()) {
    # code
  }
}
```

bnhrkn, 2022-08-04:
Nice solution. I didn't see the `operator[]` definition before. However, I think some `getDigitalNewPress` method should be added to `okapi::controller` because it would probably be friendlier to beginners coming from only PROS and should be only a few LOC. I can make a PR for this if you agree.

Octogonapus, 2022-08-04:
This sounds like a documentation problem to me. We need to make it clearer how the controller and button classes can be used. I don't want to add multiple ways to do the same thing, that will add to the confusion. Opened #490 for the docs problem.

closed 2022-08-04 by Octogonapus